peng-zhihui/HoloCubic #2: 核心板射频部分需不需要阻抗匹配

Seumi, 2021-01-23:
看了下PCB，LAN_IN引脚直接10mil连到陶瓷天线上，这个线宽2层板应该匹配不到50欧姆，这样射频性能不会变差吗

hexesdesu, 2021-01-24:
不是杠啊= =这么个小东西追求射频性能是要极限距离连接吗= =我自己用esp8266或者esp32的时候直接干上一根飞线或者干脆啥都不接都能联网= =也不知道为啥，只要不是信号特别差2.4g没那么高要求吧

peng-zhihui, 2021-01-24:
PicoD4 SiP封装内部是有匹配电路的，PCB走线只要不是太长基本没影响，而且我用的那个陶瓷天线是高增益的，实测信号很好

发送自 Windows 10 版邮件应用

发件人: hexesdesu
发送时间: 2021年1月24日 上午 8:21
收件人: peng-zhihui/HoloCubic
抄送: Subscribed
主题: Re: [peng-zhihui/HoloCubic] 核心板射频部分需不需要阻抗匹配 (#2)

不是杠啊= =这么个小东西追求射频性能是要极限距离连接吗= =我自己用esp8266或者esp32的时候直接干上一根飞线或者干脆啥都不接都能联网= =也不知道为啥，只要不是信号特别差2.4g没那么高要求吧

Raiiwhen, 2021-01-26:
陶瓷天线天线可以指下下路吗？封装名称AN5120在某宝搜不到鸭。

jiangotto, 2021-01-26:
> 陶瓷天线天线可以指下下路吗？封装名称AN5120在某宝搜不到鸭。

立创商城

dogezhou, 2021-01-26:
Nothing to preview

hexesdesu, 2021-01-27:
> 陶瓷天线天线可以指下下路吗？封装名称AN5120在某宝搜不到鸭。

淘宝寒武纪电子，夸克和这个应该都是用的c03，他们家还有另外一种双极wifi天线c01，更小

yangshengli1998, 2021-01-29:
那个环境光传感器找不到呀 具体型号可以告知一下吗

k-ying, 2021-02-02:
> 陶瓷天线天线可以指下下路吗？封装名称AN5120在某宝搜不到鸭。

可能是手滑了，RAINSUN官网相近规格的2.4GHz天线型号是AN-2051

peng-zhihui, 2021-02-02:
crossair c3

发自我的iPhone

closed 2021-02-03 by peng-zhihui