peng-zhihui/HoloCubic #13: 将MainBord上的电容C7，由 0.1uf 改为 10uf 后，可以实现Ardunio IDE下的一键下载

tokory2333, 2021-02-04:
在Ardunio IDE环境下，在c7使用0.1uf电容时，下载程序需要手动将GPIO0拉低才能在复位时进入烧写状态，不利于调试。
原因是0.1uf电容加10k电阻组成的复位电路，充电时间所需要的时间极短，复位完成时，Ardunio IDE的烧录程序来不及将CP2102的RTS拉高、DTR拉低，以至于ESP32复位结束后识别到的GPIO0依旧为3.3V，无法进入烧录模式。
通过计算，在此电路中，复位时间与RC成正比，所以加大电容或电阻可以延长复位时间。理论上1uf的就足够了。
由于手头没有1uf的电容，所以直接上了10uf，实测可以在Ardunio IDE下一键下载。其他环境由于下载程序的不同，可能会有差异。
OTA的兄弟当我没说。。。。。。

peng-zhihui, 2021-02-04:
嗯这个是有道理的，可能我的电脑配置比较好所以没遇到这个问题（大概？），不过理论上确实复位电路可以改进一下

发自我的iPhone

jsyzthz, 2021-02-06:
我的电脑试了下，10uf还是烧不进去，GPIO0拉低也没用，现在在检查焊接问题......X﹏X

YuFeiFelix, 2021-02-23:
> 我的电脑试了下，10uf还是烧不进去，GPIO0拉低也没用，现在在检查焊接问题......X﹏X

我也是X_x

fxlpop123, 2021-03-09:
ESP32 - A fatal error occurred: Failed to connect to ESP32: Timed out waiting for packet header
Arduino IDE 现在一直报错，老哥能不能提供一个思路，初步怀疑焊接问题

YuFeiFelix, 2021-03-09:
> ESP32 - A fatal error occurred: Failed to connect to ESP32: Timed out waiting for packet header
> Arduino IDE 现在一直报错，老哥能不能提供一个思路，初步怀疑焊接问题

焊接问题，我手焊了三块都是这个问题，嘉立创SMT了5块全部OK，一样的图纸。

lzx8589561, 2021-03-10:
R3电阻更换为0Ω即可实现自动下载，若不去则不稳定，容易卡死无法刷写，esp-idf环境

xingrz, 2021-06-22:
> 理论上1uf的就足够了。

正好想起来这个 issue，楼主是对的。

在[ESP32硬件设计指南](https://www.espressif.com/sites/default/files/documentation/esp32_hardware_design_guidelines_cn.pdf) V3.0 的修订里，官方也将这颗电容从 0.1uF 改成了 1uF。

[image]

[image]

apple596, 2022-04-13:
想请问一下，可以在电脑上模拟LVGL的界面，使用.bin文件吗

xingrz, 2022-04-13:
> 想请问一下，可以在电脑上模拟LVGL的界面，使用.bin文件吗

我觉得你应该在正确的地方提问