peng-zhihui/HoloCubic #19: 求如何自学整套理论和实践

YuanDdQiao, 2021-02-09:
您好大佬，我是个热爱技术的小伙，但是一直苦于时间关系，且没有方向指引。
一直想学习整一套底层硬件设备，电路，到接口，再到上层软件开发的流程。
目前只会硬件之上的软件层面，缺乏底层硬件和接口的理论和能力。
求推荐书籍，实在是内心焦灼。感谢🙏

YuanDdQiao, 2021-02-09:
如果可以的话，希望加您个微信，我不会打搅你。我只希望，得到一两个关键的建议指导。或者回复我下我去买对应的书，学习起来。看到大佬们的成果，实在是内心焦灼不安

zhang8083, 2021-02-25:
把这个项目搞起来，遇到什么学什么，遇山修路，遇水搭桥，搞完了就基本就差不多了，哈哈

NBchitu, 2021-03-22:
模电、数电，单片机，单片机与底层硬件关系比较密切，特别是数电，如SPI、I2C、CAN总线这些调试的时候都需要用到电路的知识，建议去学习下数字电路、微机原理这些大学的课程。

fan-ziqi, 2021-05-23:
> 把这个项目搞起来，遇到什么学什么，遇山修路，遇水搭桥，搞完了就基本就差不多了，哈哈

赞同 稚晖大佬说过“边打仗边学习”

debugly, 2021-11-06:
@YuanDdQiao 你好，想问问你那怎么样了，有进展了么，我是一个 iOS 开发者，刚了解到 ESP32 ，也很想学习下呢

QiangZiBro, 2021-11-09:
我两个月从零做的，现在已经做出来的，边学边做，收获很大。欢迎交流，微信：QiangZiBroWeChat

peng-zhihui, 2021-11-09:
很棒，也欢迎分享补充教程哈，我比较忙没空整理很细的～

发自我的iPhone

peng-zhihui, 2021-11-09:
> 我两个月从零做的，现在已经做出来的，边学边做，收获很大。欢迎交流，微信：QiangZiBroWeChat

很棒，也欢迎分享补充教程哈，我比较忙没空整理很细的～