peng-zhihui/HoloCubic #27: 器件选型的问题

w-wu14, 2021-03-04:
请教大佬两个器件选型的问题
1.S8050 npn晶体管电压40V的可以用吗？
2.TPYEC转USB接口用哪个型号的？
谢谢~

xusixteenqq, 2021-03-05:
1.可以
2.立创商城
商品编号： C393939

w-wu14, 2021-03-05:
> 1.可以
> 2.立创商城
> 商品编号： C393939

谢谢

w-wu14, 2021-03-16:
> C393939

TYPEC这个不对啊，难怪CP2012一直识别出来，查了几天，居然是typec接口的问题，。晕

xusixteen, 2021-03-17:
我用的这个能认串口cp2012x，但是下载程序超时，正在查原因。

w-wu14, 2021-03-17:
> 我用的这个能认串口cp2012x，但是下载程序超时，正在查原因。

信号线都不对，怎么会识别呢，你确认一下接口的地方吧，大概率是那里出问题了

xusixteen, 2021-03-17:
信号线哪里不对？
[image]
[image]

w-wu14, 2021-03-17:
你看接口的信号定义和原理图上的是不是一致

w-wu14, 2021-03-17:
你这个CP2012是在哪里买的呀

w-wu14, 2021-03-17:
> 信号线哪里不对？
> [image]
> [image]

你看一下这个typec的引脚接口和原理图以及版图上是不是一致的。另外你这个CP2012在哪买的呀

xusixteen, 2021-03-17:
我买的这家https://item.taobao.com/item.htm?spm=a1z10.3-c-s.w4002-21223910208.12.38616a4bplxhVa&id=522575454240
但是怀疑他家的ESP32有问题。我在立创商城又买了几片还没试。打样了一个开发板，把所有引脚引出来，同时加大了焊盘长度等板子到了我试一下。
[image]
[image]
只有A8和B8是反的，但是没有影响。

w-wu14, 2021-03-17:
> 我买的这家https://item.taobao.com/item.htm?spm=a1z10.3-c-s.w4002-21223910208.12.38616a4bplxhVa&id=522575454240
> 但是怀疑他家的ESP32有问题。我在立创商城又买了几片还没试。打样了一个开发板，把所有引脚引出来，同时加大了焊盘长度等板子到了我试一下。
> [image]
> [image]
> 只有A8和B8是反的，但是没有影响。

接上USBtypec接口后D+和D-的电平是多少啊，因该是一对差分信号吧？我测出来都是低电平

xusixteen, 2021-03-23:
我重新焊接了之后可以了，你也可以查一下焊接问题。淘宝定做钢网，选不带外框的小钢片，25包邮。

w-wu14, 2021-03-23:
> 我重新焊接了之后可以了，你也可以查一下焊接问题。淘宝定做钢网，选不带外框的小钢片，25包邮。

嗯嗯 我也调通了 可以显示图片 效果还是挺不错的

closed 2021-03-23 by w-wu14

AnselPeng2019, 2021-10-17:
> 我重新焊接了之后可以了，你也可以查一下焊接问题。淘宝定做钢网，选不带外框的小钢片，25包邮。

你好，可以麻烦上个链接吗？在淘宝上没找到这个

xusixteen, 2021-10-17:
3哈iUH0XtaeBvs哈 https://m.tb.cn/h.f4AHWo7?sm=889d8c  专业SMT钢网制作PCB模板贴片SMT锡膏红胶激光钢网pcb钢网37*47CM