peng-zhihui/HoloCubic #32: A fatal error occurred: Timed out waiting for packet content

Xuyinsong, 2021-03-15:
下载时经常遇到：A fatal error occurred: Timed out waiting for packet content 和 A fatal error occurred: MD5 of file does not match data in flash! 两个问题。中间也有下载成功过，不知道是什么原因导致的。请教一下应该如何解决！

diystring, 2021-03-15:
这个大概率就是 esp32 芯片没有焊接好，复位信号给不过去一直连接不上

Xuyinsong, 2021-03-16:
> 这个大概率就是 esp32 芯片没有焊接好，复位信号给不过去一直连接不上

可以了，谢谢！

closed 2021-03-16 by Xuyinsong

Niffler25, 2021-09-20:
> > 这个大概率就是 esp32 芯片没有焊接好，复位信号给不过去一直连接不上
> 
> 可以了，谢谢！

能说下最后怎么解决的吗？我也遇到了一样的问题

Xuyinsong, 2021-09-23:
重新焊了芯片

从 Windows 版邮件发送

发件人: Tractor
发送时间: 2021年9月20日 16:54
收件人: peng-zhihui/HoloCubic
抄送: Xuyinsong; State change
主题: Re: [peng-zhihui/HoloCubic] A fatal error occurred: Timed out waitingfor packet content (#32)

这个大概率就是 esp32 芯片没有焊接好，复位信号给不过去一直连接不上
可以了，谢谢！
能说下最后怎么解决的吗？我也遇到了一样的问题