peng-zhihui/HoloCubic #33: 编译一直会报这个，不知用不用管，烧录成功后重新上电没反应

diystring, 2021-03-15:
In file included from sketch\rgb_led.h:4:0,
                 from sketch\imu.h:8,
                 from C:\Users\Rtwo\Downloads\HoloCubic-main\2.Firmware\HoloCubic_fw\HoloCubic_fw.ino:2:
C:\Users\Rtwo\Documents\Arduino\libraries\FastLED/FastLED.h:14:21: note: #pragma message: FastLED version 3.003.002
 #    pragma message "FastLED version 3.003.002"
                     ^
In file included from C:\Users\Rtwo\Documents\Arduino\libraries\FastLED/FastLED.h:65:0,
                 from sketch\rgb_led.h:4,
                 from sketch\imu.h:8,
                 from C:\Users\Rtwo\Downloads\HoloCubic-main\2.Firmware\HoloCubic_fw\HoloCubic_fw.ino:2:
C:\Users\Rtwo\Documents\Arduino\libraries\FastLED/fastspi.h:130:23: note: #pragma message: No hardware SPI pins defined.  All SPI access will default to bitbanged output
 #      pragma message "No hardware SPI pins defined.  All SPI access will default to bitbanged output"

lmf-bit, 2021-03-19:
我也一样，有什么解决办法吗

w-wu14, 2021-03-20:
> In file included from sketch\rgb_led.h:4:0,
> from sketch\imu.h:8,
> from C:\Users\Rtwo\Downloads\HoloCubic-main\2.Firmware\HoloCubic_fw\HoloCubic_fw.ino:2: [...]

问题解决了吗？

diystring, 2021-03-20:
> 问题解决了吗？

没有啊，求大佬指教，vs2019、Arduino1.8.13 编译的时候都有这个，官网下载的esp32固件是能正常烧录到板子上的，或者你这边能不能编译一个二进制固件我烧录试试看 diystring@qq.com

w-wu14, 2021-03-20:
> > > In file included from sketch\rgb_led.h:4:0,
> > > from sketch\imu.h:8,
> > > from C:\Users\Rtwo\Downloads\HoloCubic-main\2.Firmware\HoloCubic_fw\HoloCubic_fw.ino:2:
> > > C:\Users\Rtwo\Documents\Arduino\libraries\FastLED/FastLED.h:14:21: note: #pragma message: FastLED version 3.003.002
> > > # pragma message "FastLED version 3.003.002"
> > > ```
> > >                  ^
> > > ```
> > > 
> > > 
> > > In file included from C:\Users\Rtwo\Documents\Arduino\libraries\FastLED/FastLED.h:65:0,
> > > from sketch\rgb_led.h:4,
> > > from sketch\imu.h:8,
> > > from C:\Users\Rtwo\Downloads\HoloCubic-main\2.Firmware\HoloCubic_fw\HoloCubic_fw.ino:2:
> > > C:\Users\Rtwo\Documents\Arduino\libraries\FastLED/fastspi.h:130:23: note: #pragma message: No hardware SPI pins defined. All SPI access will default to bitbanged output
> > > # pragma message "No hardware SPI pins defined. All SPI access will default to bitbanged output"
> > 
> > 
> > 问题解决了吗？
> 
> 没有啊，求大佬指教，vs2019、Arduino1.8.13 编译的时候都有这个，官网下载的esp32固件是能正常烧录到板子上的，或者你这边能不能编译一个二进制固件我烧录试试看 [diystring@qq.com](mailto:diystring@qq.com)

这个信息不用管，不是error

diystring, 2021-03-29:
> 这个信息不用管，不是error

Writing at 0x0000e000... (100 %)
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.0 seconds (effective 5461.4 kbit/s)...
Hash of data verified.
Compressed 18656 bytes to 12053...
Writing at 0x00001000... (100 %)
Wrote 18656 bytes (12053 compressed) at 0x00001000 in 1.1 seconds (effective 139.7 kbit/s)...
Hash of data verified.
Compressed 486432 bytes to 242697...
Writing at 0x00010000... (6 %)
Writing at 0x00014000... (13 %)
Writing at 0x00018000... (20 %)
Writing at 0x0001c000... (26 %)
Writing at 0x00020000... (33 %)
Writing at 0x00024000... (40 %)
Writing at 0x00028000... (46 %)
Writing at 0x0002c000... (53 %)
Writing at 0x00030000... (60 %)
Writing at 0x00034000... (66 %)
Writing at 0x00038000... (73 %)
Writing at 0x0003c000... (80 %)
Writing at 0x00040000... (86 %)
Writing at 0x00044000... (93 %)
Writing at 0x00048000... (100 %)
Wrote 486432 bytes (242697 compressed) at 0x00010000 in 21.4 seconds (effective 181.5 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 128...
Writing at 0x00008000... (100 %)
Wrote 3072 bytes (128 compressed) at 0x00008000 in 0.0 seconds (effective 1228.8 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...

是的，不用管也能烧录成功，但是不知道为啥，烧录成功后重新上电不能正常运行，RGB没反应，屏幕有背光无图像，感觉程序没有运行

w-wu14, 2021-03-29:
> 是的，不用管也能烧录成功，但是不知道为啥，烧录成功后重新上电不能正常运行，RGB没反应，屏幕有背光无图像，感觉程序没有运行

LED没反应可能是没有焊接好，先确保各个管脚焊接OK，如果还没有反应，用示波器看一下输入输入管脚有没有数据输入。另外，要显示图像，你先确保SD卡可以正常读写，然后再看能否显示图片。

syj0925, 2021-04-30:
LED灯没亮，应该是程序阻塞在mpu.init这里面，检查一下MPU的电路。