peng-zhihui/HoloCubic #52: MPU6050中央焊盘是否应该接地？

fan-ziqi, 2021-05-23:
稚晖君您好，我发现您的PCB中MPU6050中央焊盘为GND网络且打了过孔，但MPU6050数据手册中说明
“The MPU-60X0 has very low active and standby current consumption. The exposed die pad is not required for heat sinking, and should not be soldered to the PCB. Failure to adhere to this rule can induce performance changes due to package thermo-mechanical stress. There is no electrical connection between the pad and the CMOS.”
MPU-60X0有很低的动态性能和标准的电流损耗。该焊盘不要求进行散热，同时也不能连接到PCB上。该焊盘是为了增加芯片的机械应力，不应该有任何的电气连接。“
请问您将中央焊盘接地是什么原因，接地会不会对MPU6050造成影响？如果手册中说不接地应该怎么处理呢？

peng-zhihui, 2021-05-23:
是的，这个是我之前没注意，但是实测没太大影响，你可以自己把pcb相关连线去掉一下。

发自我的iPhone

fan-ziqi, 2021-05-23:
> 是的，这个是我之前没注意，但是实测没太大影响，你可以自己把pcb相关连线去掉一下。 发自我的iPhone
> […](#)

多谢大佬回复~这就去打一版试一下

fan-ziqi, 2021-05-23:
多谢大佬回复~这就去打一版试一下

 
 
 
 
 
 
 
 
 
 
   
   
&mdash;

closed 2021-06-08 by peng-zhihui

franva, 2021-07-17:
@fan-ziqi  @peng-zhihui  硬件方面小白问一个问题。

请问去哪里打板子呢？ 能不能给个链接？ 多谢啦～！