peng-zhihui/HoloCubic #53: 向稚辉君案例开源项目 - LightCube

Charmve, 2021-06-11:
# LightCube 💠<br>A 3D Dynamic Display System Based on Voice Control
<p align="left">
<a href="https://charmve.github.io/doc/papers/一种基于语音控制的3D动态显示系统设计.pdf" target="_blank">[Paper (Chinese)]</a> | <a href="https://charmve.github.io/doc/src/作品海报Poster.pdf" target="_blank">[Poster]</a> | <a href="https://charmve.github.io/doc/src/基于FPGA的智能语音3D动态显示系统·Slides.pdf" target="_blank">[Slides]</a> | <a href="https://charmve.github.io/doc/src/基于FPGA的智能语音3D动态显示系统·Slides.pdf" target="_blank">[Patents]</a>
</p>

[image]

<p align="center">[image]  [image]</p>

- 项目地址：https://github.com/Charmve/LightCube

- 项目简介 (**100** 字以内)：

光立方作为一种新型的裸眼3D显示技术，用户不需要配戴任何观看辅助工具就能达到裸眼3D显示的效果，给人们带来了全新的视觉体验。本项目设计一种基于语音控制的3D动态显示系统，采用LD3320非特定人声语音识别芯片以及STM32F407作为光立方核心控制器实现全彩高阶的语音控制光立方设计，解决了现有光立方显示色彩单一、分辨率低、人机交互性能差、设计复杂、成本较高等的问题。语音识别模块与STM32之间采用UART串口通信，将语音识别结果作为控制指令发送给STM32，控制光立方的显示动画和工作模式，并用模块自带的喇叭播放背景音乐。SM16126串转并级联驱动输出电路的设计，降低了系统功耗，为高阶光立方的设计提供了一种可行的实现方案。测试表明，光立方显示系统对语音指令识别的响应速度快、识别精度高、工作稳定，具有较好的使用体验，使人们的生活更加智能化和人性化。

- 项目截图

<p align="center">
  [image]
  <br>(<b>click</b> this picture to watch a video)
  <br><br>[image]
  <br>Fig 1. System block
</p>

<h2></h2>

- **衍生项目**

[image]

# EmotionCube 💠 <br> Intelligent Speech Companion Robot Based-on Sentiment Analysis

<p align="left">
  [image]
  [image]
  [image]
  [image]
  [image]
</p>
<br>

以情绪分析和智能语音为研究对象，作者充分调研了目前家居机器人的研究现状，大部分都是以释放人类体力与脑力为目标，而忽视了“家”的真正含义；在这个基础上作者提出了智能家居的新方向——情感交互。整个项目的最终目标是为了营造一个真正的智能“家”居，让科技与人类情感产生共鸣，缓解人们的生活压力，打造一个温馨舒适的情感家居环境。采用全彩12 阶光立方作为三维动态显示器，基于表情识别和智能语音制作一种智能陪伴机器人。

本系统主要由三部分组成：
- 表情识别摄像头——树莓派+摄像头搭建完成；
- 智能音箱——FPGA 语音识别和麦克风播报；
- 三维动态显示系统——包括电源开关电路、控制电路和级联驱动电路，并留下足够空间扩展后续电路。

全彩的3D 显示效果增加了使用者的视觉体验，表情识别和智能语音促进了人机交互，真正实现了人机语音交互，更加智能化、人性化。

<div align=center>[image]</div>
<p align="center"><b>Emotion Cube</b></p>

franva, 2021-07-17:
好像除了用来看以外 没什么其他用处啊？

Charmve, 2021-07-19:
> 好像除了用来看以外 没什么其他用处啊？

它本质上就是一个三维阵列

nekocharm, 2021-09-19:
光立方除了占地方还是占地方

dengqianggh, 2022-09-05:
确实有点占地方，出租屋装不下 了

gregrgr, 2022-10-07:
大二自己做了一个，真的无聊