peng-zhihui/HoloCubic #58: 购买三棱镜贴膜问题

franva, 2021-07-17:
硬件方面的小白完全体有个问题：

我准备跟着Gihub 下手试一试这个项目啦。 

想问一下 我看到了2个棱镜：
1. https://item.taobao.com/item.htm?id=608974731299&ali_trackid=2:mm_26632674_99800204_108708850297:1626497619_140_1019591071&union_lens=lensId:TAPI@1626497450@210561ad_093c_17ab2cd80be_9f8f@01;linkRuleId:2;recoveryid:1626497619_140_1019591071&spm=a21wu.12321156.0.0&ak=26008535&bxsign=tbkwC6LH/YYinyaaGXJRpDwtLRN3HuN2LFz/ag0N6vGbZJn695xCDoYEuiyMTCTuPy/8i5hOZytTTDsa4rTR5HiUtSegwUK+X45Det/tIEXJ8g=

2. https://item.taobao.com/item.htm?spm=a21wu.12321156.0.0.d54270ee4aMdYQ&id=611950292711

第一个它说没有镀膜， 是不是就是不能买？
[image]
我记得 zhihui君在视频里提到过 两块棱镜的粘合面是镀有分光膜的。所以没有镀膜就是没有分光膜 对吗？

然后 我看我即使选择不同的颜色 比如 黑色、米白色、灰色等 淘宝网站上现实的都是透明的三棱镜。
所以它这个颜色不是说三棱镜玻璃体的颜色，而是说 镀膜的颜色？？？
我一直以为一定要选 透明的 才行？

没想到在第一步 就卡住了。 毕竟不在国内 买错了一次 估计要等好几周到1个多月才能再次收到了，所以十分小心。

希望zhihuijun和各位大佬能帮忙排忧解难一下。 非常感谢～！

Bollod, 2021-11-24:
请问，购买三棱镜需要镀膜吗？

franva, 2021-11-27:
@Bollod  不需要 , 我已经买过, 验证过了. 只要三棱镜就行. @lkShen

peng-zhihui, 2021-11-27:
不是买三棱镜，是买分光棱镜

franva, 2021-11-29:
[image]

多谢@peng-zhihui 解释。  
不过我已经买过了。 当时搜索的是 三棱镜，反正目前看是可以的。 估计是 歪打正着 买成了 分光镜吧。

Bollod, 2021-11-29:
> [image]
> 
> 多谢@peng-zhihui 解释。 不过我已经买过了。 当时搜索的是 三棱镜，反正目前看是可以的。 估计是 歪打正着 买成了 分光镜吧。

哈哈，我直接在淘宝买了现成的分光棱镜，已经粘合好的。不过按照我对分光棱镜原理理解，是不是单独的一块分光棱镜就可以进行反射，不需要两块重叠一起。

franva, 2021-11-29:
@Bollod  对的。 我上面的照片就能看出啦， 只需要一个 三棱镜就行。 我想 志辉君是为了美观吧。

能分享一下你的那个淘宝链接吗？

Bollod, 2021-11-29:
> @Bollod 对的。 我上面的照片就能看出啦， 只需要一个 三棱镜就行。 我想 志辉君是为了美观吧。
> 
> 能分享一下你的那个淘宝链接吗？

`38信就而来个他去在过地时起啊 https://m.tb.cn/h.fi6YTyR?sm=e57178  分光棱镜 25.4mm实验教学仪器 半透半反1:1分光 影像分光投射棱镜`

我理解不只是美观，两块在一起会有透视效果，单独的一块只是反射，也就是从背面可以看到镜像。

franva, 2021-11-29:
@Bollod  多谢分享。

我2个放在一起试过。没有透视效果。 

你的已经是2个在一起了， 有透视效果吗？ 拍一张照片 看看 透视效果到底是什么意思， 语言描述 估计说不清。

Bollod, 2021-11-29:
> 

还在家扔着，下班我看看。方便的话可以加V，我是这方面新手，不太懂底层这些。目前第一步想用树莓派作样例出来，再用其它板子试试。。。急需一位硬件大佬指点，目前线都不知道怎么接😂

franva, 2021-11-29:
可以加VX, 我也是在琢磨： franva