peng-zhihui/HoloCubic #59: OCA贴膜问题

franva, 2021-07-24:
请问大家， 在这个项目里面是不是要用到OCA两次在两个不同的地方？

1.用OCA把两个三棱镜粘起来
2.用OCA把这2个棱镜组成的方块粘到屏幕上？

如果是，那你们是怎么裁减OCA贴膜的呢？ 市面上买的贴膜肯定都比1.3存屏幕和三棱镜的贴合面大。
如果直接用剪刀裁减 会不会把OCA贴膜的边界给挤压 的不平整了？

PetyC, 2021-08-23:
直接网购成品的分光棱镜

franva, 2021-08-23:
@PetyC  多谢

请问 能提供一个链接吗?
谢谢

woflyrun, 2021-09-06:
> @PetyC 多谢
> 
> 请问 能提供一个链接吗?
> 谢谢

just in China,  Alibaba , taobao

woflyrun, 2021-09-06:
> @PetyC 多谢
> 
> 请问 能提供一个链接吗?
> 谢谢

可以搜索， 分光棱镜， 1：1

woflyrun, 2021-09-06:
> @PetyC 多谢
> 
> 请问 能提供一个链接吗?
> 谢谢

https://s.click.taobao.com/RL5yMiu

PetyC, 2021-09-06:
> @PetyC
> 
> 请问 能提供一个链接吗?

https://item.taobao.com/item.htm?spm=a1z09.2.0.0.7b362e8dCj4Z2l&id=583924549232&_u=62ih642n716e

franva, 2021-09-06:
多谢二位: @woflyrun @PetyC 

我已经买过了, 买了2块三棱镜, 又买了OCA, 没想到还比这个贵 $_$

closed 2021-09-06 by franva

woflyrun, 2021-09-06:
> 多谢二位: @woflyrun @PetyC
> 
> 我已经买过了, 买了2块三棱镜, 又买了OCA, 没想到还比这个贵 $_$

自己黏贴的好像不行吧，这个是镀了一层分光膜，不然的话两块三棱镜就是全反射，不会有透视效果的， 1:1分光棱镜是50%反射，50% 透射， 这样就会有背景透视效果，当然你自己贴的也是可以用，但是没有透视效果，就像是一面镜子一样，这是我的观点，实际效果要等你做好才知道。

franva, 2021-09-18:
我来更新啦~~  根本不需要什么膜.  就是把2个三棱镜放在一起就行了, 亲测有效 :)