peng-zhihui/HoloCubic #62: 请教一下，MainBoard和ExtendBoard之间的连接端子是哪一款

Niffler25, 2021-08-23:
请教一下，MainBoard和ExtendBoard之间的连接端子是哪一款。双排2.54端子针太大，不合适。

woflyrun, 2021-09-07:
2.54的不合适吗， 我刚买的，还没到
[image]

Niffler25, 2021-09-20:
> 2.54的不合适吗， 我刚买的，还没到
> [image]

你这个大小合适吗？可以发下链接吗？

woflyrun, 2021-09-22:
> > 2.54的不合适吗， 我刚买的，还没到
> > [image]
> 
> 你这个大小合适吗？可以发下链接吗？

https://item.taobao.com/item.htm?spm=a1z09.2.0.0.4b102e8dpmmPhp&id=624202093978&_u=m26tdfv4e297
我买的是单个4P的， 两个放两个  ，感觉大小正好，但是我板子焊接没成功， 目前正在找问题，这个尺寸我看着没问题

perseconds, 2023-04-07:
用这个弹簧顶针为什么屏幕只有背景光，不显示效果啊