peng-zhihui/HoloCubic #84: 刷入固件后串口一直输出rst:0x10 (rtcwdt_rtc_reset),boot:0x13 (spi_fast_flash_boot)

mrhp02, 2022-08-25:
烧录显示成功，但是串口输出为以下内容，怎么解决？
rst:0x10 (rtcwdt_rtc_reset),boot:0x13 (spi_fast_flash_boot) configsip: 188777542, spiwp:0xee clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00 mode:dio, clock div:1 load:0x3fff0018,len:4 load:0x3fff001c,len:1216 ho 0 tail 12 room 4 load:0x40078000,len:10944 load:0x40080400,len:6388 entry 0x400806b4

[image]

dengqianggh, 2022-09-05:
我也遇到了，同样的问题，可以谷歌一下，据说是芯片供电问题，或者是稚晖君所说的那个引脚问题。

mrhp02, 2022-09-05:
谢谢，我再试试---- 回复的原邮件 ***@***.***>发送日期2022年09月06日 00:06 ***@***.***> ***@***.******@***.***>主题Re: [peng-zhihui/HoloCubic] 刷入固件后串口一直输出rst:0x10 (rtcwdt_rtc_reset),boot:0x13 (spi_fast_flash_boot) (Issue #84)

我也遇到了，同样的问题，可以谷歌一下，据说是芯片供电问题，或者是稚晖君所说的那个引脚问题。

[

{

"target": "https://github.com/peng-zhihui/HoloCubic/issues/84#issuecomment-1237257385",

"url": "https://github.com/peng-zhihui/HoloCubic/issues/84#issuecomment-1237257385",

"name": "View Issue"

},

"description": "View this Issue on GitHub",

"publisher": {

"name": "GitHub",

"url": "https://github.com"

}

}

]

SunRich, 2022-09-18:
> 我也遇到了，同样的问题，可以谷歌一下，据说是芯片供电问题，或者是稚晖君所说的那个引脚问题。

能指导下吗？现在无法烧录新的固件了~~

mrhp02, 2022-10-11:
我也没找到办法，重新焊了。---- 回复的原邮件 ***@***.***>发送日期2022年09月18日 09:49 ***@***.***> ***@***.******@***.***>主题Re: [peng-zhihui/HoloCubic] 刷入固件后串口一直输出rst:0x10 (rtcwdt_rtc_reset),boot:0x13 (spi_fast_flash_boot) (Issue #84)

我也遇到了，同样的问题，可以谷歌一下，据说是芯片供电问题，或者是稚晖君所说的那个引脚问题。

能指导下吗？现在无法烧录新的固件了~~

[

{

"target": "https://github.com/peng-zhihui/HoloCubic/issues/84#issuecomment-1250172715",

"url": "https://github.com/peng-zhihui/HoloCubic/issues/84#issuecomment-1250172715",

"name": "View Issue"

},

"description": "View this Issue on GitHub",

"publisher": {

"name": "GitHub",

"url": "https://github.com"

}

}

]

QQ3516969, 2022-12-02:
不省心啊，屏幕接的vspi，卡模块接的hspi
[image]