peng-zhihui/HoloCubic #85: 无法上传固件，报错：A fatal error occurred: Failed to connect to ESP32: No serial data received.

ZFYoung, 2022-09-04:
Retrieving maximum program size .pio\build\pico32\firmware.elf
Checking size .pio\build\pico32\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [==        ]  17.7% (used 57924 bytes from 327680 bytes)
Flash: [====      ]  39.7% (used 520333 bytes from 1310720 bytes)
Configuring upload protocol...
AVAILABLE: cmsis-dap, esp-bridge, esp-prog, espota, esptool, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa
CURRENT: upload_protocol = esptool
Looking for upload port...
Using manually specified: COM4
Uploading .pio\build\pico32\firmware.bin
esptool.py v3.3
Serial port COM4
Connecting......................................

A fatal error occurred: Failed to connect to ESP32: No serial data received.
For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html

固件上传失败，尝试GPIO0短接GND试过了也是这个报错失败，重新焊接后依旧。有人知道怎么解决吗

ZFYoung, 2022-09-04:
设备管理器里端口是存在的。 驱动更新的官网CP210x_Universal_Windows_Driver.zip版本

ZFYoung, 2022-09-04:
esp32虚焊导致， 重新焊接后ok了

closed 2022-09-04 by ZFYoung

AzureRubia-Guet, 2023-09-28:
老哥你这个真的是帮了我大忙，我esp32-wroom-32u也有这个问题，找了半天，最后是按照你的短接gpio0和gnd解决的